Neotron-Compute/Neotron-32-Hardware #7: Render drawings in CI

thejpster, 2019-12-16:
Kicad Automation task hangs waiting for window to open.

eeshow renders schematic but all components are missing. Have contacted author.

IGBC, 2019-12-16:
Maybe automating DRC and ERC in CI is also worth doing?

thejpster, 2019-12-16:
Agreed. Title changed.

The kicad automation can do DRC, Gerber's, ERC and pdf plots. The eeshow tool is an independent renderer of eeschema drawings, so won't do ERC or DRC or Gerber's, but it doesn't involve running Kicad in a headless X server and doing screen scraping.

thejpster, 2019-12-21:
Looks like it might be a Docker/X issue. If I run Docker with the --user argument on my machine I get the same issue GitHub Actions shows.

I at least have a release script now so I can rattle off ERC, DRC and all the plots with a one liner.

thejpster, 2021-06-18:
See Neotron Pico for an example of how to do this.